ros/joint_state_publisher #4: Races in GUI thread during joint state update

disRecord, 2017-10-19:
This problem occurs when GUI is running and `source_list` topic is used to update joint state. Pose is updated only partially for long JointState messages.

`source_cb()` callback emits `sliderUpdateTrigger` signal. But it seems during sliders' values update in [`update_sliders()`](https://github.com/ros/joint_state_publisher/blob/f4305f7895ba74f39b46e1357ac4ca0931990e23/joint_state_publisher/joint_state_publisher/joint_state_publisher#L397) somehow [`onValueChange()`](https://github.com/ros/joint_state_publisher/blob/f4305f7895ba74f39b46e1357ac4ca0931990e23/joint_state_publisher/joint_state_publisher/joint_state_publisher#L385) is called. So pose is updated only partially: half of the pose comes from received `source_list` message and other half comes from the old sliders' state.

Attached patch adds a lock and solves this problem.
[joint_state_publisher.patch.txt](https://github.com/ros/joint_state_publisher/files/1398164/joint_state_publisher.patch.txt)

clalancette, 2017-10-19:
Would you mind opening a pull request with that change?  It makes it easier for us to review and track it.  Thanks.

disRecord, 2017-10-19:
I've created pull request: https://github.com/ros/joint_state_publisher/pull/5 .

v4hn, 2018-05-25:
This bug is a huge pain for users of MoveIt!'s demo mode,
where the joint_state_publisher is used to simulate the robot driver.

If users enable the GUI via a parameter in the `demo.launch` file, it looks like everything is running correctly.
But when they execute planned trajectories only some of the planned joints, if any, actually move.
All others remain at their current positions, resulting in broken and very unintuitive trajectories.

A number of hours of debugging by multiple users went into this already,
it would be great to see this fixed one way or another.

sloretz, 2019-10-15:
Was fixed by #11 :tada:, closing

closed 2019-10-15 by sloretz