ros/joint_state_publisher #12: Use dynamic reconfigure instead of standalone gui?

lucasw, 2018-08-05:
The dynamic reconfigure option could use https://github.com/pal-robotics/ddynamic_reconfigure_python .

If the user wanted a standalone gui they could use https://github.com/lucasw/dynamic_reconfigure_tools/tree/master/rqt_dr_single (which I could split out into a standalone repo and actually release it, or maybe merge it with dynamic_reconfigure).

The use_gui flag could either be deprecated and the user would have to independently run a dynamic reconfigure client (which means I don't need to move sooner on releasing rqt_dr_single)- or an instance of rqt_dr_single could be launched programmatically?

Randomize and Center would become bool_t and would uncheck themselves after being checked.

Some of the gui configurability would be lost but this repo would be simplified, the qt dependency eliminated.

Another option would be to convert this to an rqt plugin and preserve the existing ui exactly, but have it be able to be embedded within a larger rqt panel or launched standalone.  Like the dynamice reconfigure case would be simpler to get rid of the use_gui flag than preserve it.

lucasw, 2018-08-05:
TODO If use_gui is false, don't launch the dr server.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)

closed 2025-06-02 by sloretz