ros/joint_state_publisher #20: ros2-devel  Parameter not declared exception

Jconn, 2019-07-16:
when I run this on dashing, passing my urdf file to the node, I get an exception:

rclpy.exceptions.ParameterNotDeclaredException: ('Invalid access to undeclared parameter(s)', 'dependent_joints')

I modified my the JointStatePublisher:get_param to catch that exception and return None, and the node works now.

clalancette, 2019-07-17:
Hm, yeah, I haven't tried the port on dashing yet.  We really need to change the code to `declare_parameter` before we use it.  I probably won't have time for that soon, but if you provide a PR I can review it.  Thanks.

clalancette, 2020-03-10:
This should be fixed by the port that was just done in #30.  Closing, but feel free to reopen if you still have problems.

closed 2020-03-10 by clalancette