ros/joint_state_publisher #33: remapping via <launch>.py and logging is not working as expected

mganglb, 2019-12-20:
Hello ros2 developers,
I am quite new to ros2 (Eloquent), so maybe i just use the functionalities wrong. (sry for that in advance)

# logging
for some reason i dont see the logging on "info" level. chanaging the logging level does not solve the problem. Logging on "warn" level works for me.

`
  node = rclpy.create_node('joint_state_publisher', allow_undeclared_parameters=True,
                             automatically_declare_parameters_from_overrides=True)
  my_logger = node.get_logger()
  my_logger.set_level(LoggingSeverity.WARN) # seems to have to influence anyway
  my_logger.warn("this works")              # visible in console
  my_logger.info('this works not)           # not visible in console
`

# remapping
If i start up the node via <launch>.py launc-file the remapping does not work for me. But mostlikely 
i just use it worng.
`    remappings_robot_description=[((namespace, '/robot_description'), '/robot_description')]`
`
    start_joint_state_publisher_cmd = Node(
        ...,
        remappings=remappings_robot_description)
`

clalancette, 2019-12-20:
For general ROS 2 questions, please first try https://answers.ros.org.  It's often the case that the question has already been asked and answered there, and it serves as a central repository for user information about ROS 2.  Given that, I'll close this out, but feel free to keep commenting or re-open if you think this is in error.

closed 2019-12-20 by clalancette

mganglb, 2019-12-20:
remapping works fine btw, was my own mistake