ros/joint_state_publisher #36: AttributeError: JointStatePublisher instance has no attribute 'source_update_cb'

igzubov, 2020-02-24:
Getting this error on Ubuntu 16.04 and ROS Kinetic:
`[ERROR] [1582565767.970762]: bad callback: <bound method JointStatePublisher.source_cb of <joint_state_publisher.JointStatePublisher instance at 0x7ff0c19dc5a8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/joint_state_publisher/__init__.py", line 202, in source_cb
    if self.source_update_cb is not None:
AttributeError: JointStatePublisher instance has no attribute 'source_update_cb'
`

Why `source_update_cb` is set to None aftrer subscribers are created? Shouldn't this variable be initialized before creating subscribers?

sloretz, 2020-03-10:
@igzubov thanks for the report! Would you mind trying #41 on your system and reporting if it fixes the issue?

igzubov, 2020-03-12:
@sloretz tested, fixes.

sloretz, 2020-03-12:
Thanks @igzubov , closing as fixed by #41 and #43

PR #41 (2020-03-11): [kinetic] Set source_update_cb attr before creating subscribers
Fixes #36. It's possible for a subscriber to receive their first message and call the callback prior to `self.source_update_cb` existing. This PR creates the attribute before the subscribers.
closed 2020-03-12 by sloretz