ros/joint_state_publisher #47: Is joint_state_publisher working in dashing?

ManuelZ, 2020-04-12:
I'm trying to go through the [URDF tutorial](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Movable%20Robot%20Model%20with%20URDF) but in ROS2 and I'm getting this error when launching `joint_state_publisher`:

```
$ ros2 run joint_state_publisher joint_state_publisher

Traceback (most recent call last):
  File "/opt/ros/dashing/lib/joint_state_publisher/joint_state_publisher", line 11, in <module>
    load_entry_point('joint-state-publisher==2.0.0', 'console_scripts', 'joint_state_publisher')()
  File "/opt/ros/dashing/lib/python3.6/site-packages/joint_state_publisher/joint_state_publisher.py", line 415, in main
    jsp = JointStatePublisher(parsed_args.urdf_file)
  File "/opt/ros/dashing/lib/python3.6/site-packages/joint_state_publisher/joint_state_publisher.py", line 234, in __init__
    self.dependent_joints = self.parse_dependent_joints()
  File "/opt/ros/dashing/lib/python3.6/site-packages/joint_state_publisher/joint_state_publisher.py", line 174, in parse_dependent_joints
    dependent_joints = self.get_parameters_by_prefix('dependent_joints')
AttributeError: 'JointStatePublisher' object has no attribute 'get_parameters_by_prefix'
```

According to [issue 30](https://github.com/ros/joint_state_publisher/pull/30), this should already be working in ROS2. I'm using Dashing on Ubuntu 18.04.

This is part of my launch file:
```
    return LaunchDescription([
        Node(package='robot_state_publisher',
             node_executable='robot_state_publisher',
             output='screen',
              arguments=[urdf_filename]
        ),

        Node(package='joint_state_publisher',
             node_executable='joint_state_publisher',
             output='screen'
        ),

        Node(package='rviz2',
             node_executable='rviz2',
             condition=IfCondition(LaunchConfiguration('rviz'))
        ),
    ])

Am I doing something wrong?
```

ManuelZ, 2020-04-12:
I've noticed that the method "get_parameters_by_prefix" from rclpy is not present in the dashing branch in [rclpy.node.Node](https://github.com/ros2/rclpy/blob/dashing/rclpy/rclpy/node.py)

clalancette, 2020-04-13:
> I've noticed that the method "get_parameters_by_prefix" from `rclpy` is not present in the dashing branch in [rclpy.node.Node](https://github.com/ros2/rclpy/blob/dashing/rclpy/rclpy/node.py)

I really thought I had tested this on Dashing, but you are correct; `get_parameters_by_prefix` doesn't exist in Dashing rclpy.  I *think* we can emulate this by using `get_parameters` instead; I'll give that a shot.

ManuelZ, 2020-04-14:
Let me know if I can be of help.

clalancette, 2020-04-14:
> Let me know if I can be of help.

I'll take you up on that :).  Yesterday I was able to figure out how to emulate `get_parameters_by_prefix` just by iterating over `self._parameters`; the changes in the latest commit on https://github.com/ros/joint_state_publisher/tree/dashing-fixes have that fix.

What I have not yet been able to figure out is how to run the launch tests (in the test/ subdirectory) on Dashing.  Dashing did not yet have hooks for pytest, so it won't automagically run the launch tests.  If you have time to look into that and propose a solution, that would be great.  Just open a PR that targets the dashing-fixes branch when you have something.

And thanks for the help!

ManuelZ, 2020-04-15:
Ok, I'll tackle it.

ManuelZ, 2020-05-02:
Sorry I don't think I'll be able to have a solution for this.

clalancette, 2020-05-19:
All right, no worries.

What I think I'm going to do here is:

1.  Separate the dashing and eloquent branches.
1.  On the dashing branch, disable the tests.
1.  Also on the dashing branch, make the small fixes needed to make this work on dashing.
1.  Re-release for both eloquent and dashing.

clalancette, 2020-05-28:
I released the updated version for Dashing, with the fixes and without the tests.  So I'll close this out.  Thanks for the report.

closed 2020-05-28 by clalancette