ros/joint_state_publisher #49: ImportError: No module named PyQt5.QtCore and PySide2.QtCore

Luca8991, 2020-05-11:
in my _.launch_ file I have this line:
`<node pkg="joint_state_publisher_gui" type="joint_state_publisher_gui" name="jnt_st_pub" />`

here is the error after calling my _.launch_ file:

```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/joint_state_publisher_gui/joint_state_publisher_gui", line 41, in <module>
    from python_qt_binding.QtWidgets import QApplication
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 276, in <module>
    getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 120, in _select_qt_binding
    (', '.join(["'%s'" % b for b in binding_order]), '\n'.join(error_msgs)))
ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):
  ImportError for 'pyqt': No module named PyQt5.QtCore
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 153, in _load_pyqt
    _named_import('PyQt5.%s' % module_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ImportError: No module named PyQt5.QtCore

  ImportError for 'pyside': No module named PySide2.QtCore
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 185, in _load_pyside
    _named_import('PySide2.%s' % module_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ImportError: No module named PySide2.QtCore
```

aguadopd, 2022-01-18:
@Luca8991 could you fix it?

Luca8991, 2022-01-19:
> @Luca8991 could you fix it?

I honestly don't remember how I managed to solve it

closed 2025-06-02 by sloretz