ros/joint_state_publisher #56: AttributeError: 'module' object has no attribute 'JointStatePublisher'

mn-banjar, 2020-10-14:
Hi there,
I'm using the joint state publisher to control a robot in rviz

I got this error:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 44, in <module>
    jsp = joint_state_publisher.JointStatePublisher()
AttributeError: 'module' object has no attribute 'JointStatePublisher'
[joint_state_publisher-4] process has died [pid 5590, exit code 1, cmd /opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher __name:=joint_state_publisher __log:=/home/ros/.ros/log/6ab3238a-0e39-11eb-a164-0800278c4f52/joint_state_publisher-4.log].
log file: /home/ros/.ros/log/6ab3238a-0e39-11eb-a164-0800278c4f52/joint_state_publisher-4*.log
[ERROR] [1602692558.462153185]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?

```

sloretz, 2020-10-15:
> By the way, I tested the package and launch file on another machine and it works fine

Mind posting the output of `apt show ros-kinetic-joint-state-publisher` on the machine that it works, and on the machine that it doesn't work?

clalancette, 2021-01-04:
Closing due to no response.  Feel free to reopen if you are still having this problem.

closed 2021-01-04 by clalancette