ros/joint_state_publisher #62: failes to colcon build

newcanopies, 2021-04-26:
Hi, I am not able to build JSP on Foxy/Ubuntu20.04 -- any clues?

joint_state_publisher/joint_state_publisher$ colcon build

```
Starting >>> joint_state_publisher
--- stderr: joint_state_publisher
CMake Error: The current CMakeCache.txt directory /home/user/colcon_ws/src/joint_state_publisher/joint_state_publisher/build/joint_state_publisher/CMakeCache.txt is different than the directory /home/user/moon/joint_state_publisher/joint_state_publisher/build/joint_state_publisher where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/user/moon/colcon_ws/src/joint_state_publisher/joint_state_publisher/CMakeLists.txt" does not match the source "/home/user/moon/joint_state_publisher/joint_state_publisher/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
---
Failed   <<< joint_state_publisher [0.12s, exited with code 1]

Summary: 0 packages finished [0.47s]
  1 package failed: joint_state_publisher
  1 package had stderr output: joint_state_publisher
```

sloretz, 2021-04-26:
The error message says CMake can't find `catkin`, but `catkin` is only available in ROS 1. Did you check out the the default branch? Checkout the `foxy` branch for building on Foxy instead.

```
git clone https://github.com/ros/joint_state_publisher.git -b foxy
```

You might be interested in the documentation for building ROS Foxy from source: https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html

If more issues come up, please ask for help on https://answers.ros.org per the [support guidelines](https://docs.ros.org/en/foxy/Contact.html#support).

closed 2021-04-26 by sloretz

newcanopies, 2021-04-26:
forgot to source Foxy branch 
`$ git checkout foxy`

newcanopies, 2021-04-26:
thanks @sloretz 
realized this moments ago at viewing: CMakeLists.txt:4 `find_package(catkin REQUIRED)`