ros/joint_state_publisher #65: Misleading docs for 'zeros' parameter for ROS2

hello-binit, 2021-09-21:
Hello, I'm on Ubuntu 20.04, using ROS2 Galactic. Reading the joint_state_publisher [README](https://github.com/ros/joint_state_publisher/blob/ros2/joint_state_publisher/README.md), I'd expect the 'zeros' parameter to take a dictionary, mapping strings to floats. In a ROS2 launch file, I'd expect this to be:

```python
joint_state_publisher_gui_node = Node(
    package='joint_state_publisher_gui',
    executable='joint_state_publisher_gui',
    parameters=[{'zeros': {'joint_lift': 0.2, 'joint_wrist_yaw': 3.4}}]
)
```

The snippet above doesn't work. Looking at the code, 'zeros' is retrieved using the [`get_parameters_by_prefix`](https://github.com/ros/joint_state_publisher/blob/foxy/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L238) method. So the following snippet works instead:

```python
joint_state_publisher_gui_node = Node(
    package='joint_state_publisher_gui',
    executable='joint_state_publisher_gui',
    parameters=[{'zeros.joint_lift': 0.2, 'zeros.joint_wrist_yaw': 3.4}]
)
```

Let me know if I've misunderstood anything. If it is an issue, happy to file a PR improving the README.

clalancette, 2021-09-22:
> Let me know if I've misunderstood anything. If it is an issue, happy to file a PR improving the README.

You've understood it perfectly.  I was trying to get this across in this sentence:

```
In Eloquent, pass on the command-line as '-p zeros.joint_name:=value'.
```

But if you think the wording could be further improved, please feel free to open a PR.  I'm happy to review it.

DLu, 2021-09-28:
Good to close?

PR #66 (2021-09-28): Clarify docs on 'mapped parameters' in README
This PR rewords the README to clarify that the parameters `zeros` and `dependent_joints` don't actually take in dictionaries, but instead follow the `<parameter>.<key>:=<value>` format. It also points out that `ros2 param list` won't show these two params, but that they are still supported. This PR closes #65.

If this change to the README looks good, I can copy it to a wiki. Is there a ROS2 equivalent for the [joint_state_publisher ROS wiki](http://wiki.ros.org/joint_state_publisher)?
closed 2021-09-28 by clalancette