ros/joint_state_publisher #75: Can i move my robot arm like rviz?

EmadDeve20, 2021-12-08:
My question sounds confusing I want my robot to move exactly like the robot inside rviz not like your code!
Is this possible? I know I mean badly right now, so let me explain more precisely!

 look at picture in down

[image]

now look at this picture

[image]

As you can see, I used your code in the second image, but it was different from the first image! The difference is that I want to shake that orange robot using a code like yours!

I saw your rviz message and your code was no different:

[image]
^ this message from RVIZ

[image]
^ this message from your UI

Is there a way I can shake like a rviz (orange robot) using a slider like your code? Your code is like doing a plan, but I want it before! Is there a way? I'm really confused.

clalancette, 2021-12-08:
For questions like this, please ask them on https://answers.ros.org, which is our central question and answer site.  You should be able to get some help there.  I'm going to close this one out.

closed 2021-12-08 by clalancette

EmadDeve20, 2021-12-09:
@clalancette Thanks :D