ros/joint_state_publisher #82: ParameterAlreadyDeclaredException on initialization with use_sim_time

xouillet, 2022-06-03:
Hi,

On galactic, with last version of `rclpy` (1.9.1), `joint_state_publisher` fails to initialize if we set the parameter `use_sim_time`.

```
Traceback (most recent call last):   File "/opt/ros/galactic/lib/joint_state_publisher/joint_state_publisher", line 11, in <module>
     load_entry_point('joint-state-publisher==2.2.0', 'console_scripts', 'joint_state_publisher')()   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher/joint_state_publisher.py", line 415, in main
     jsp = JointStatePublisher(parsed_args.urdf_file)   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher/joint_state_publisher.py", line 217, in __init__
     super().__init__('joint_state_publisher', automatically_declare_parameters_from_overrides=True)
   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 216, in __init__     self.declare_parameters(
   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 487, in declare_parameters     raise ParameterAlreadyDeclaredException(parameters_already_declared)
rclpy.exceptions.ParameterAlreadyDeclaredException: ('Parameter(s) already declared', ['use_sim_time'])
```

That's because `rclpy` v. 1.9.1 introduces this commit: https://github.com/ros2/rclpy/commit/4d97b4ecf0ae5ae42a4ed7e41cffcacb51a85219 that now raises `ParameterAlreadyDeclaredException` when using `automatically_declare_parameters_from_overrides`.

Basic fix could be to set `automatically_declare_parameters_from_overrides` to `False`, as I'm not sure this is necessary.

feixyz10, 2022-06-14:
Same problem was encountered.

Kaetera, 2022-06-20:
Hello, same for us on Galactic.
This is preventing to simulate our system in Gazebo

xouillet, 2022-07-01:
Actually, issue has been spotted and fixed in `rclpy`, cf. https://github.com/ros2/rclpy/pull/896. Issue should be fixed with the new release of rclpy 1.9.2 released yesterday.

clalancette, 2022-11-04:
Given that this was fixed in rclpy, going to close this out.

closed 2022-11-04 by clalancette