ros/joint_state_publisher #85: Gracefully handle SIGINT

zflat, 2022-10-10:
I noticed a similar issue in the `joint_state_publisher` that was fixed in https://github.com/RobotWebTools/rosbridge_suite/pull/794/files regarding graceful shutdown of the python node.

Should the same fix be applied? Which is to move the calls to `jsp.destroy_node()` and `rclpy.shutdown()` to be within the `try` block and not after it?

The error I currently get is:

> [joint_state_publisher-5] Traceback (most recent call last):
[joint_state_publisher-5]   File "/opt/ros/humble/lib/joint_state_publisher/joint_state_publisher", line 33, in <module>
[joint_state_publisher-5]     sys.exit(load_entry_point('joint-state-publisher==2.3.0', 'console_scripts', 'joint_state_publisher')())
[joint_state_publisher-5]   File "/opt/ros/humble/lib/python3.10/site-packages/joint_state_publisher/joint_state_publisher.py", line 423, in main
[joint_state_publisher-5]     rclpy.shutdown()
[joint_state_publisher-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 126, in shutdown
[joint_state_publisher-5]     _shutdown(context=context)
[joint_state_publisher-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/utilities.py", line 58, in shutdown
[joint_state_publisher-5]     return context.shutdown()
[joint_state_publisher-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/context.py", line 102, in shutdown
[joint_state_publisher-5]     self.__context.shutdown()
[joint_state_publisher-5] rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:241

zflat, 2022-10-10:
We could also put in a check before calling shutdown like this:

```python
    if rclpy.utilities.ok():
        rclpy.shutdown()
```

mateusz-lichota, 2022-12-16:
I don't think this check will ever go through, cause rclpy.spin() spins until the context is shutdown, so it will never be ok() once we exit from spin()

zflat, 2022-12-16:
> I don't think this check will ever go through, cause rclpy.spin() spins until the context is shutdown, so it will never be ok() once we exit from spin()

Makes sense looking at https://github.com/ros2/rclpy/blob/c3a679666cce40379dadd96f9b8b4c7c237e0b5f/rclpy/rclpy/__init__.py#L221

Then the fix of moving the calls to `jsp.destroy_node()` and `rclpy.shutdown()` to be within the try block seems to be the simplest.

zflat, 2022-12-16:
Addressed in pull request #86

zflat, 2022-12-16:
Resolved in https://github.com/ros/joint_state_publisher/pull/86

closed 2022-12-16 by zflat