ros/joint_state_publisher #87: Incorrect order of joints using ROS Humble

IvoD1998, 2023-02-15:
Hello

When trying to use the joint states of the robot for a forward kinematic calculation, we noticed that the order of the joint positions seems to be published incorrectly. 
It appears that the order of joints is now published as:

- joint1
- joint2
- joint5
- joint3
- joint4
- joint6

(see screenshot below for added clarity)

[image]

Thank you in advance!
Ivo Dekker

clalancette, 2023-02-15:
Can you please describe why that is wrong, and how joint_state_publisher is involved here?

IvoD1998, 2023-02-15:
We were wondering why the list of joints as published by the joint state publisher does not match the sequential order they are supposed to be in according to the robot URDF. We tested this with various different robot models and got the same result.
It could of course also be an issue on the side of the MoveIt configs for the robots, but we are uncertain.

karanchahal, 2023-03-05:
Can I use foxglove to do this ? What do I need to do ?

sloretz, 2023-03-06:
It looks like there is an intent to publish the joints in a fixed order:

https://github.com/ros/joint_state_publisher/blob/424fd9e1955fddf38f1b3e1721baf9f6e9322d6b/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L344

Mind posting a `urdf` that shows the issue? It might be the change in order came from the urdf parser, as that's where `joint_state_publisher` gets the order from

https://github.com/ros/joint_state_publisher/blob/424fd9e1955fddf38f1b3e1721baf9f6e9322d6b/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L87-L98

antrom99, 2025-02-27:
@IvoD1998 

I also have the same problem in ros2 humble; the order of links and joints in the **robot.model.xacro** file goes from base_link to joint6 and despite this on /joint_states I have this order:

```
header:
stamp:
sec: 1740586545
nanosec: 900974358
frame_id: ''
name:
- joint2
- joint3
- joint1
- joint4
- joint5
- joint6
position:
- 0.0
- -1.57
- 0.0
- 0.0
- -1.57
- 0.0
velocity:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0

effort:
- .nan
- .nan
- .nan
- .nan
- .nan

```

did you solve the problem? If so, how?

IvoD1998, 2025-02-28:
Hello @antrom99 

My apologies for being inactive on this matter. I have since been preoccupied with other matters.
However, we solved this by applying a filter ourselves which puts the joint data in the correct order again. 
In my opinion this could be good practice anyways, as it is always wise to assume that the order of the data is not the exact structure you expect it to be. 
In short, we created a map where the joint is the key and the second map elements contains a custom struct which holds the pos, vel and effort data for each joint. After this, we can simply access the desired joint via the map key.

Hope this helps!

closed 2025-02-28 by IvoD1998

antrom99, 2025-03-01:
@IvoD1998 

Thanks for the answer, i resloved the issue with a little change to the **joint_state_broadcaster***