ros/joint_state_publisher #99: joint_state_publisher not running propperly

anthiv, 2023-09-05:
Hello
I have a problem when running this command "roslaunch mobile_robot rviz.launch" and I think it is related to joint_state_publisher. Here is my launch file:

<?xml version="1.0"?>
<launch>

    <param name="robot_description" command="xacro '$(find mobile_robot)/urdf/rover.urdf.xacro'"/>
    <arg name="rviz_config_file" default="$(find mobile_robot)/config/robot.rviz"/>

    <node name="joint_state_publisher_gui" pkg="joint_state_publisher_gui" type="joint_state_publisher_gui"/> 
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_config_file)"/>

</launch>

and although I don't get any errors, the preview of my robot is not correct (I use 4 wheels in my urdf file and I only get one wheel in the center of my robot car as pictured below). When I run "rosnode list" I can only see /robot_state_publisher and not /joint_state_publisher_gui and I can't figure why. Also when I kill the procedure with Ctrl+C I get this :
[joint_state_publisher_gui-2] killing on exit
[joint_state_publisher_gui-2] escalating to SIGTERM
[joint_state_publisher_gui-2] escalating to SIGKILL
[rosout-1] killing on exit
[master] killing on exit
Shutdown errors:
 * process[joint_state_publisher_gui-2, pid 26370]: required SIGKILL. May still be running.
 
which is also related to joint_state_publisher_gui. I tried to use joint_state_publisher but I came across the same issue. My code works fine in my friend's laptop and we have installed ros the same way. Do you have any idea where the problem stands?
[image]

yhyhhhhhh, 2023-11-19:
Hello, have you solved this problem? i met the same issue too

sean7401, 2024-09-19:
hi, i met the same issue too.......... and need help!!! plz read this comment!

closed 2025-06-02 by sloretz