ros/joint_state_publisher #105: joint_state_publisher_gui crashes on using revolute joints

Amronos, 2024-12-01:
I have defined my revolute joint in SDF like this:
```xml
    <joint name="test_joint" type="revolute">
      <parent>base_link</parent>
      <child>test_link</child>
      <pose relative_to="base_link">1 1 1 0 0 0</pose>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <lower>-inf</lower>
          <upper>inf</upper>
        </limit>
      </axis>
    </joint>
```
I get the following errors:
```
[joint_state_publisher_gui-1] [INFO] [1733051122.369655512] [joint_state_publisher_gui]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-1] [INFO] [1733051122.387361215] [joint_state_publisher_gui]: Centering
[joint_state_publisher_gui-1] [INFO] [1733051122.495230202] [joint_state_publisher_gui]: Centering
[joint_state_publisher_gui-1] Traceback (most recent call last):
[joint_state_publisher_gui-1]   File "/opt/ros/jazzy/lib/python3.12/site-packages/joint_state_publisher_gui/joint_state_publisher_gui.py", line 207, in initializeSliders
[joint_state_publisher_gui-1]     self.centerEvent(None)
[joint_state_publisher_gui-1]   File "/opt/ros/jazzy/lib/python3.12/site-packages/joint_state_publisher_gui/joint_state_publisher_gui.py", line 245, in centerEvent
[joint_state_publisher_gui-1]     joint_info['slider'].setValue(self.valueToSlider(joint['zero'], joint))
[joint_state_publisher_gui-1]                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[joint_state_publisher_gui-1]   File "/opt/ros/jazzy/lib/python3.12/site-packages/joint_state_publisher_gui/joint_state_publisher_gui.py", line 255, in valueToSlider
[joint_state_publisher_gui-1]     return int((value - joint['min']) * float(RANGE) / (joint['max'] - joint['min']))
[joint_state_publisher_gui-1]            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[joint_state_publisher_gui-1] ValueError: cannot convert float NaN to integer
[rviz2-3] [ERROR] [1733051135.281374250] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "test_link" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
[rviz2-3] [ERROR] [1733051135.281509904] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "test_link" from authority "Authority undetectable" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
```

I think the infinite values for the limit cause this. I can not use a continuous joint in this case as Gazebo does not support it.

DLu, 2024-12-10:
To me, this is a Gazebo bug, and the correct answer is to use a continuous joint. What happens in Gazebo?

Amronos, 2024-12-11:
> To me, this is a Gazebo bug, and the correct answer is to use a continuous joint. What happens in Gazebo?

https://github.com/gazebosim/sdformat/issues/326 does provide some details on this.